pavelrevak/pystlink #3: It can't work with error '*** Error unlocking FLASH ***'

chenwq3, 2017-03-09:
Hi, I trying flash stm32 device but no using Keil.
I Try it successfully on Ubuntu.

For Windows, I run into some error:

```cmd
c:\Users\chen>py Documents\pystlink\pystlink.py flash:erase flash:xpfw.bin
DEVICE: ST-Link/V2 V2J21S4
SUPPLY: 3.26V
CORE:   CortexM0
MCU:    STM32F042x6/STM32F048x6/STM32F070x6
FLASH:  32KB
SRAM:   6KB
*** Error unlocking FLASH ***
```

but when I try it again with `-d` param:

```
pystlink.py flash:erase flash:xpfw.bin  -d > stdout.txt 2> stderr.txt
```

nothing in stdout.txt, and here is the content of stderr.txt :
```
Successfully connected to ST-Link/V2
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 25 44 83 04 48 37
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
  USB > f2 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < ef 05 00 00 12 08 00 00
  USB > f2 30 a3 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 77 14 b1 0b
DEVICE: ST-Link/V2 V2J21S4
SUPPLY: 3.26V
COREID: 0bb11477
  USB > f2 36 00 ed 00 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 c2 0c 41
CPUID:  410cc200
CORE:   CortexM0
  USB > f2 36 00 58 01 40 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 45 64 00 10
IDCODE: 10006445
  USB > f2 36 cc f7 ff 1f 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 20 00 ff ff
MCU:    STM32F042x6/STM32F048x6/STM32F070x6
FLASH:  32KB
SRAM:   6KB
CMD: flash:erase
Stm32FP.flash_erase_all()
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < ef 05 00 00 15 08 00 00
Stm32.core_reset_halt()
  USB > f2 35 f0 ed 00 e0 03 00 5f a0 00 00 00 00 00 00
  USB < 80 00
  USB > f2 35 fc ed 00 e0 01 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 35 0c ed 00 e0 04 00 fa 05 00 00 00 00 00 00
  USB < 80 00
  USB > f2 36 0c ed 00 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 00 00 00
  USB > f2 36 10 20 02 40 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 00 00 00
  USB > f2 36 10 20 02 40 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 80 00 00 00
*** Error unlocking FLASH ***
Stm32.core_nodebug()
  USB > f2 35 f0 ed 00 e0 00 00 5f a0 00 00 00 00 00 00
  USB < 80 00
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
  USB > f2 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
DONE in 0.17s
```

```cmd
<... ...>
  USB < 80 00 00 00 20 00 00 00
  USB > f2 35 0c 20 02 40 20 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 35 10 20 02 40 80 00 00 00 00 00 00 00 00 00
  USB < 80 00
Stm32.core_reset_halt()
  USB > f2 35 f0 ed 00 e0 03 00 5f a0 00 00 00 00 00 00
  USB < 80 00
  USB > f2 35 fc ed 00 e0 01 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 35 0c ed 00 e0 04 00 fa 05 00 00 00 00 00 00
  USB < 80 00
  USB > f2 36 0c ed 00 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 00 05 fa
Writing FLASH: [========================================] done in 1.45s
Stm32.core_nodebug()
  USB > f2 35 f0 ed 00 e0 00 00 5f a0 00 00 00 00 00 00
  USB < 80 00
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
  USB > f2 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DONE in 1.75s
```

[image]

pavelrevak, 2017-03-14:
hello, please try first update firmware in ST-Link, on my configurations it is OK (Linux and Windows7)
my version of ST-Link is: V2J27S6

chenwq3, 2017-03-20:
I updated the firmware of ST-Link debuger, and the problem still exist.
The version of firmeware of ST-LINK debuger is: V2.J28.S7

pavelrevak, 2017-03-20:
OK, if on Ubuntu is flashing successful, can you send me also log from Ubuntu and then again from Windows on same micro-controller with same flashing command for comparison? And be sure that you have same version of pystlink. (Please send logs as attachment if it is possible). Thanks.

chenwq3, 2017-04-01:
I tested and I found that :
- If I flash the chip on a SMT Test Socket(https://www.adafruit.com/product/1280), nothing go wrong.
- If I flash the chip (that is mounted in a PCB) from the PCB's SWDIO port, And I get that errors.
> Anyway, I can use Keil to flash the chip from the PCB's SWDIO port and get no warning.

I will update if i get more information.

chenwq3, 2017-05-02:
For now, No more information.
I will update if I have useful info.
Close for now.

closed 2017-05-02 by chenwq3

pavelrevak, 2017-05-02:
probably is necessary to FLASH under reset, because the SWDIO pins are used by application and this feature is not implemented in pystlink right now.