pavelrevak/pystlink #4: Wont run

zaidshakil, 2017-06-20:
Traceback (most recent call last):
  File "pystlink.py", line 4, in <module>
    import lib.stlinkusb
  File "/home/zaidshakil/TU Berlin/Robotics/pystlink/lib/stlinkusb.py", line 1, in <module>
    import usb.core
ImportError: No module named 'usb'

i keep getting this error, please suggest. I have updated the firmware

pavelrevak, 2017-06-20:
Hello zadshakil, you don't have installed pyusb and libusb
https://github.com/pavelrevak/pystlink/blob/master/README.md#requirements

zaidshakil, 2017-06-20:
Thank you for replying. 
I have those 2 installed.

pavelrevak, 2017-06-21:
then pyusb is probably installed for different version of python

try run: `python3` and inside type `import usb.core` to test if is installed correctly.

jimfred, 2019-06-28:
I had a similar issue, was seeing 'ImportError: No module named usb.core' when running 'pystlink_test.py' from the Windows command line. When running 'python pystlink_test.py', it ran successfully.