pavelrevak/pystlink #5: *** ST-Link/V2 is not connected ***

fahad80, 2017-07-11:
I think I am missing something obvious. My ST-Link/V2-1 can be detected by the ST-Link Utility. What are the possible reasons I am having this issue?

pavelrevak, 2017-07-11:
Hello Fahad, thanks for reporting, but please provide more informations like what OS do you have, version of python, if you have all drivers installed, some debug info `-d` and so. Thanks.

fahad80, 2017-07-12:
I am using windows 10, python 3.6. I installed pyusb. Downloaded libusb from [here ](https://sourceforge.net/projects/libusb/files/)(your link isn't working) and copied libusb-1.0.dll into Windows/System32 directory. `-d` doen't provide anything apart from the "*** ST-Link/V2 is not connected ***" error.

Hope this helps.

pavelrevak, 2017-07-12:
You also need installed [ST-Link utility](http://www.st.com/en/embedded-software/stsw-link004.html) this utility also contain driver for ST-Link and this is necessary for running under windows, I looking that I forgot to mention about this in README.

fahad80, 2017-07-12:
I have that installed. I mentioned in my first post that ST-Link Utility can detect the ST-Link/V2.

ZhouChenghong, 2018-04-25:
I meet the same issue.And I use Windows10 and Python3.5.I used to use ST-Link to program SM32.I cloned pystlink and libusb from github,and compiled libusb-1.0.dll with VS.I am confused ablout what I did wrong.

ZhouChenghong, 2018-04-26:
I know why it is fail to connect to ST-Link/V2,because ST-Link/V2 use STM32 drive program instead of libusb drive program,we should change the drive program to libusb.I program flash successfully several times but failed many times with "*** Error unlocking FLASH ***".I haven't figured out it.

ZhouChenghong, 2018-05-12:
I have figured out the error. :-)  I add a 100ms delay before unlock the flash,and try to unlock the flash again between eraseing and flashing.After that,it work well.

BilgehanKosem, 2019-03-14:
> I know why it is fail to connect to ST-Link/V2,because ST-Link/V2 use STM32 drive program instead of libusb drive program,we should change the drive program to libusb.I program flash successfully several times but failed many times with "*** Error unlocking FLASH ***".I haven't figured out it.

I am facing the same problem(*** ST-Link/V2 is not connected ***). How did you handle this problem, can you explain it a little bir more detailed.

ZhouChenghong, 2019-04-17:
> I am facing the same problem(*** ST-Link/V2 is not connected ***). How did you handle this problem, can you explain it a little bir more detailed.

In fact,I almost forget what I did.Emmm,you can use Zadig to replace the driver of st-link easlly.

desowin, 2019-12-19:
There are two potential solutions. One is to use Zadig to install libusb supported driver instead of the official ST-Link driver. The other option is to reverse engineer the interface the ST-Link driver provides and support it in this library (as extra option, not replacing libusb).

SEGGER STLinkReflash uses the official ST-Link driver so the interface is either documented somewhere (would save the reverse engineering) or available upon request (my request for the raw USB protocol format used by ST-Link has been rejected, so most likely the driver interface would be rejected as well).

desowin, 2019-12-19:
Apparently the ST-Link official driver (STSW-LINK009) is just WinUSB. There's no need to replace the driver with Zadig, as it is compatible with libusb (STSW-LINK009 version 2.0.1; libusb version 1.0.22). The libusb-1.0.dll does not need to be copied to Windows directory. Copying libusb-1.0.dll to the Python installation directory (same directory where python.exe is located) is enough.

pavelrevak, 2019-12-19:
ST-Link driver is needed only to have right .inf file for USB Device.
All what you need on Windows is libusb1.dll and pyusb (I prefer 64bit python and also libusb1)

AlexandreHDK, 2020-07-24:
I was also having the same problem. It didn't work with Python 3.8.x or 3.7.x, only if I changed the driver to libusb with Zadig.
With Python 3.6.8 I don't need to change the driver.

fbvasconcelos, 2022-12-15:
In 2022 the problems continue to happen. Does anyone have a solution for this?

jguembe, 2023-01-19:
I had the same problem and I managed to solve it. In my case, I have 

- Windows 11
-  Python 3.10
-  The STM32 drivers were already installed since I usually use STM32CubeIDE to program NUCLEOS using ST-LINK/V2. 
 
At first, I tried to use the pystlink program and it did not detect any STLINK (*** ST-Link/V2 is not connected ***).
First, I tried downloading and copying libusb-1.0.dll to path Windows/System32 but it didn't work for me.

 To make it recognize, I had to load the libusb-win32 (v1.2.6.0) driver that comes by default to install with the Zadig program (https://zadig.akeo.ie/). Using this program, I selected the ST-LINK Debug device and replaced the driver. Now the python works. 

[image]

However, **the programmer of the STM32CubeIDE does not work**, you should go back to the previous version of the driver for it to work again. You can do this by reverting to the previous driver in Device Manager > select device > Driver. Or you can also do: Device Manager > select device (right click) > Update Driver > Browse my PC for drivers > Choose from a list of available drivers > Choose STMicroelectronics STLink dongle.

kRasul, 2024-07-14:
@jguembe
[image]
I accidently found the way to make it work.
It was working on my latop (with both pystlink and cubemxprogrammer) and didn't work on my PC. 
Then I installed libusb-win32 on my and got it working with pystlink but cubemxprogrammer stoopped to work. 
I found that on my laptop was older ST-link driver then on PC. Its properties on image above. Too lazy to continue hastle though since working only with pystlink is enough for me for now.