pavelrevak/pystlink #6: *** USB Error: [Errno 32] Pipe error ***

rfb, 2017-08-15:
Freshly updated STLink2

```
[45228.161514] usb 1-2.4: new full-speed USB device number 15 using xhci_hcd
[45228.266791] usb 1-2.4: New USB device found, idVendor=0483, idProduct=3748
[45228.266793] usb 1-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[45228.266794] usb 1-2.4: Product: STM32 STLink
[45228.266795] usb 1-2.4: Manufacturer: STMicroelectronics
[45228.266797] usb 1-2.4: SerialNumber: UÿlI\xffffffc2\xffffff80IU7\xffffffc2\xffffff87
```

```
> python3 pystlink.py  -d
Successfully connected to ST-Link/V2
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 27 07 83 04 48 37
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 01 00
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < ed 05 00 00 ca 08 00 00
  USB > f2 43 01 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 30 a3 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 09 00
  USB > f2 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00
*** USB Error: [Errno 32] Pipe error ***
```

rfb, 2017-08-15:
adding retry and tout options to the xfer call in read_coreid has no effect

jxltom, 2017-12-12:
Call ```Pystlink()._connector._dev.reset()``` may help...

hugokernel, 2018-08-31:
Hi,
Same problem here.
Freshly updated to V2J31S7 and I have the same error.

`Successfully connected to ST-Link/V2
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 27 c7 83 04 48 37
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 01 00
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < d9 05 00 00 fb 07 00 00
  USB > f2 43 01 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 30 a3 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 09 00
  USB > f2 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00
*** USB Error: [Errno 32] Pipe error ***
`

No problem with OpenOCD.