pavelrevak/pystlink #10: Would it be possible to extend this program to access STM8's?

AokiAhishatsu, 2018-05-18:
Im getting this (first time):
`*** USB Error: [Errno 32] Pipe error ***`
and then this (every next time)
`*** USB Error: [Errno 110] Operation timed out ***`

markfink, 2018-11-24:
same here. I updated the stlink programmer firmware... flashing stm8s is not supported (below)?
can I talk to stm8s via SWIM?

```
$ ../pystlink/pystlink -v -c STM8S103F3 flash:erase:verify:0x08000000:./blink_led.ihx 
Successfully connected to ST-Link/V2
DEVICE: ST-Link/V2 V2J32S7
SUPPLY: 3.23V
COREID: 00000042
*** Not connected to CPU ***
DONE in 0.03s
```

AokiAhishatsu, 2018-11-25:
I used [this](https://github.com/vdudouyt/stm8flash). Worked fine.

markfink, 2018-11-26:
thanks @AokiAhishatsu I have seen that but I am looking for a python package