pavelrevak/pystlink #11: pystlink as a library

simonnilsson, 2018-10-06:
Hi, great work on this implementation.

I'm in need to use this as a library in a python program I'm working on. I was able to take the lib folder, rename it to pystlink and change some imports from `lib.` to `pystlink.` and make it work as a library. 

I see that pip install support is in the planed features. I think it would be a good idea to split the project up into two parts **pystlink** (library) and **pystlink-cli** (this application). That way pystlink-cli could just have a dependency on the library. I belive the library will need some improvemnets to make it easier to use and so on but I think this would be a good first start.

I want to avoid forking the project and doing it myself as I belive it's is better to focus on improving the original project. Thoughts?

pavelrevak, 2018-10-06:
Hi simonnilsson,
here is what you want https://github.com/cortexm/pyswd

simonnilsson, 2018-10-06:
Hi Pavel, I actually tried that library aswell but kept getting a 

`USB Error: [Errno 10060] Operation timed out` 

I think the error happens when using read_mem() with a (size > 4) and (size % 4 != 0) 

I didn't have that problem using the code in this repository that is why I choose this instead. But if you think pyswd is the way to go then I will head over to that repo and report this issue there. 
`

closed 2018-10-07 by pavelrevak

sabarikrishna4, 2025-04-09:
Hello Both,
I am trying flash the STM32 using python, While searching I found that pystlink in GitHub, unfortunately i can't able to pip install it. May i know how to it?