pavelrevak/pystlink #17: STM32FP - direct flashing is reverted

pavelrevak, 2019-10-17:
@UweBonnes make great work, but now I see that his patch don't work with STM32F0 (I don't make tests with other FP MCUs)
Currently is the code reverted to flashing using microcode.
Also I don't understand why this e72aca9 ? because this stop currently running program in MCU but don't restart all the peripherals and it make many problems.

UweBonnes, 2019-10-17:
I tested successfull with on a F07_discovery :+1: 
f07_discovery> pystlink flash:erase:verify:gpio.bin 
DEVICE: ST-Link/V2-1 V2J35M26
SUPPLY: 3.26V
CORE:   CortexM0
MCU:    STM32F070xB/STM32F071xB/STM32F072xB/STM32F078xB
FLASH:  128KB
SRAM:   16KB
Loaded 13964 Bytes from gpio.bin file
Erasing FLASH: [========================================] done in 0.18s
Writing FLASH: [========================================] done in 0.41s
Verify FLASH : [========================================] done in 0.09s

After flashing the F07_disco works as expected.

Can you be more verbose?

UweBonnes, 2019-10-17:
Regarding e72aca9: For pure flashing we can always cold-plug (connect under reset). Cold plug is also needed if the running program remaps the SWD pins or if the program sleep/stop/ etc for a long time.  I only had flashing in mind. This needs to be rethought...

pavelrevak, 2019-10-18:
...
Writing FLASH
enable pgbit
  USB > f2 35 10 20 02 40 01 00 00 00 00 00 00 00 00 00
  USB < 80 00
Writing FLASH: [                                        ]   0%
  USB > f2 48 00 00 00 08 00 04 00 00 00 00 00 00 00 00
  USB > 00 10 00 20 4d 26 00 08 41 19 00 08 3d 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 41 19 00 08 2d 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 c9 18 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 3f 19 00 08 30 b5 05 00 00 23 2f 35 2c 78 e4 b2 9c 42 2c d1 01 33 2b 70 f0 25 3b 33 83 85 43 68 02 62 44 62 5d 60 83 68 1c 60 04 68 d9 60 21 00 28 31 99 60 5a 60 91 22 1a 60 03 6a 19 00 fe 2b 00 d9 ff 21 5b 1a 03 62 18 00 42 1e 90 41 78 22 c9 b2 09 04 0a 43 59 42 4b 41 59 06 13 00 80 22 00 06 03 43 0b 43 92 01 13 43 63 60 01 23 18 00 30 bd 10 b5 05 4b 18 68 02 00 2f 32 13 78 00 2b fc d1 03 4a 03 49 ff f7 bd ff 10 bd d4 00 00 20 01 02 00 00 e3 00 00 20 00 29 02 d0 43 7a 00 2b 05 d1 42 72 00 2a 03 d0 00 23 03 72 03 60 70 47 00 29 07 d0 03 68 0b 4a 0a 33 03 60 93 42 01 d9 01 23 83 72 03 7a 8b 42 f1 d0 01 72 00 29 01 d0 00 23 eb e7 03 4b 02 68 9a 42 f9 d8 43 68 01 33 43 60 e4 e7 e7 03 00 00 94 46 f0 b5 11 4c 9b b2 25 88 9d 42 06 d0 00 2d 19 d0 65 88 02 35 6d 00 64 19 f5 e7 65 88 7f 29 0f dc 04 34 6f 18 8f 42 0b d0 80 29 09 d0 62 46 23 88 8e 00 93 40 82 59 01 31 13 43 83 51 02 34 f1 e7 a8 1c f0 bd 28 00 fc e7 c0 46 b8 39 00 08 f8 b5 06 00 17 00 1d 00 0c 00 2b 78 00 2b 07 d0 21 00 3a 00 30 00 ff f7 cb ff 01 35 24 18 f4 e7 20 00 f8 bd 70 b5 04 00 00 20 0a 4a 0a 4d 13 88 ab 42 0e d0 53 78 16 88 5b 06 36 05 5b 0f 36 0d a6 42 02 d1 8b 42 00 d1 10 00 01 33 5b 00 d2 18 ed e7 70 bd 00 7c 00 08 ff ff 00 00 10 b5 00 21 ff f7 e0 ff 00 28 01 d0 40 78 c0 09 10 bd 43 68 58 69 80 07 c0 0f 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 02 22 00 29 01 d0 1a 83 70 47 5a 83 fc e7 02 22 43 68 1a 83 70 47 02 22 43 68 5a 83 70 47 43 68 18 69 01 23 18 40 70 47 43 68 58 69 01 23 18 40 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 01 22 00 29 01 d0 1a 83 70 47 5a 83 fc e7 01 22 43 68 1a 83 70 47 01 22 43 68 5a 83 70 47 43 68 18 69 80 07 c0 0f 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 18 69 00 07 c0 0f 70 47 43 68 58 69 00 07 c0 0f 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 08 22 00 29 01 d0 1a 83 70 47 5a 83 fc e7 08 22 43 68 1a 83 70 47 08 22 43 68 5a 83 70 47 43 68 58 69 80 06 c0 0f 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 20 22 00 29 01 d0 1a 83 70 47 5a 83 fc e7 20 22 43 68 1a 83 70 47 20 22 43 68 5a 83 70 47 43 68 58 69 c0 06 c0 0f 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 43 68 10 22 00 29 01 d0 1a 83 70 47 5a 83 fc e7 10 22 43 68 1a 83 70 47 10 22 43 68 5a 83 70 47 03 68 70 b5 9d 68 1b 69 04 00 98 47 01 21 41 40 c9 b2 20 00 a8 47 70 bd 30 b5 8a 18 84 68
Writing FLASH: [==                                      ]   6%
  USB > f2 48 00 04 00 08 00 04 00 00 00 00 00 00 00 00
  USB > 53 1a 00 2b 0a dd c3 68 00 2b 07 d0 01 3b c3 60 0b 78 65 1c 85 60 01 31 23 70 f0 e7 00 23 23 70 30 bd c3 68 00 2b 08 d0 01 3b c3 60 83 68 5a 1c 82 60 00 22 19 70 83 68 1a 70 70 47 30 b5 8a 18 c4 68 53 1a 00 2b 0a dd 03 69 00 2b 07 d0 01 3b 03 61 0b 78 65 1c c5 60 01 31 23 70 f0 e7 00 23 23 70 30 bd 03 69 00 2b 08 d0 01 3b 03 61 c3 68 5a 1c c2 60 00 22 19 70 c3 68 1a 70 70 47 30 b5 8a 18 c4 68 53 1a 00 2b 0a dd 03 69 00 2b 07 d0 01 3b 03 61 0b 78 65 1c c5 60 01 31 23 70 f0 e7 00 23 23 70 30 bd 03 69 00 2b 08 d0 01 3b 03 61 c3 68 5a 1c c2 60 00 22 19 70 c3 68 1a 70 70 47 30 b5 8a 18 84 69 53 1a 00 2b 0a dd c3 69 00 2b 07 d0 01 3b c3 61 0b 78 65 1c 85 61 01 31 23 70 f0 e7 00 23 23 70 30 bd c3 69 00 2b 08 d0 01 3b c3 61 83 69 5a 1c 82 61 00 22 19 70 83 69 1a 70 70 47 f8 b5 05 00 0c 00 17 00 5e 18 00 2b 03 da 5b 42 71 1c ff f7 f5 ff 01 22 ba 40 b4 42 07 d0 7f 2c 05 dc a0 00 2b 58 01 34 13 43 2b 50 f5 e7 f8 bd f7 b5 0f 27 03 69 01 91 8c 33 1e 68 05 00 77 43 0e 49 38 00 02 f0 5d f9 04 1e 01 d1 00 2e 10 dc 0b 4b 9f 42 00 dd 0f 24 00 26 b4 42 0b dd 72 00 52 42 1f 32 03 23 01 99 e8 68 ff f7 c9 ff 01 36 f3 e7 01 24 f0 e7 f7 bd c0 46 40 9c 00 00 ff c3 09 00 f0 b5 85 b0 02 92 01 93 01 9b 1c 78 00 2c 41 d0 01 33 01 93 0a 9b 08 33 1e 68 a6 42 06 d0 00 2e 36 d0 5a 68 02 32 92 00 9b 18 f5 e7 0a 9a 5e 68 52 68 03 92 7f 29 29 dc 08 33 00 2e 16 da 0c 00 9c 46 35 00 01 35 63 46 22 00 01 2d 11 d1 ac 18 a4 46 94 45 1a d0 80 2a 18 d0 10 cb 02 9f 95 00 bc 40 47 59 01 32 3c 43 44 51 f2 e7 35 00 0a 00 ed e7 00 2c eb dd 67 46 08 cf a2 00 bc 46 02 9f 01 3c bb 40 87 58 3b 43 83 50 db e7 03 9b 9e 19 89 19 b9 e7 08 00 05 b0 f0 bd f7 b5 08 9e b4 1c 25 78 9d 42 05 d0 00 2d 26 d0 65 78 02 35 64 19 f6 e7 73 78 65 78 00 93 7f 29 1a dc 1f 2a 18 dc 63 1a 54 42 a4 46 02 33 6f 18 01 93 8f 42 10 d0 80 29 0e d0 01 9b 5e 5c 34 00 94 40 00 2a 02 da 34 00 66 46 34 41 8e 00 83 59 01 31 1c 43 84 51 ec e7 00 9b 58 19 fe bd 28 00 fc e7 f7 b5 06 00 17 00 1d 00 0c 00 2b 78 00 2b 09 d0 08 9a 21 00 00 92 30 00 3a 00 ff f7 c0 ff 01 35 24 18 f2 e7 20 00 fe bd 43 68 18 69 c0 06 c0 0f 70 47 43  [... cut ...]  07 ae 04 33 a2 69 08 96 09 93 d7 7a 03 91 00 2f 22 d0 09 22 15 49 50 43 02 f0 20 f8 00 23 01 00 00 93 20 31 20 33 03 22 05 a8 ff f7 8f ff 33 00 2a 00 03 99 e0 68 ff f7 09 fd a3 69 01 00 da 7a 0b 4b 00 93 0b 4b 00 2a 00 d1 0b 4b 2a 00 e0 68 ff f7 42 ff 0b b0 f0 bd
*** USB Error: [Errno 60] Operation timed out ***
Stm32.core_nodebug()
  USB > f2 35 f0 ed 00 e0 00 00 5f a0 00 00 00 00 00 00
*** USB Error: [Errno 60] Operation timed out ***
DONE in 1.03s

I see this with china programmer ST-Link V2

pavelrevak, 2019-10-18:
ou, I probably see that, from witch version ST-LINK API has STLINK_DEBUG_APIV2_WRITEMEM_16BIT functions?

UweBonnes, 2019-10-18:
If I remember right from V2.28. It took probably until V2.32 until things stabilized. In blackmagic I have following code:
        if ((Stlink.ver_stlink < 3 && Stlink.ver_jtag < 32) ||
                (Stlink.ver_stlink == 3 && Stlink.ver_jtag < 3)) {
                DEBUG("Please update Firmware\n");
                goto error_1;
        }

B.t.w. did you try the pc-stlinkv2 platform of the blackmagic debug probe with your clone?

UweBonnes, 2019-10-18:
Argh, I typed 21 instead of 32! Twice of by one on the keyboard.
https://github.com/pavelrevak/pystlink/pull/18

UweBonnes, 2019-10-19:
Did the Stlink upgrade work for you? Did flashing work with the upgraded probe?

UweBonnes, 2019-10-19:
And before upgrade, did pystlink report the need for upgrade?

pavelrevak, 2019-10-19:
Yes, it was reported (I miss this info, I take some old STLINK device) now after upgrade it is working also with direct flashing!
sorry this was my fault..
right now, only problem I see is reset_halt vs halt only on running firmware before flashing. is necessary to restart program and keep it halted before flashing.

pavelrevak, 2019-10-19:
I just reverted my revert

UweBonnes, 2019-10-19:
The intention is following for flashing:
- unconditionally reset the device, so we catch devices with SWD remaped or deep sleeping (cold plug)
- Setup device to halt on reset
- release reset so we get halted
- Flash it
- remove halt on reset
- Pulse reset for a clean start of the device if -r is not given

For the other commands the CPU state is important. A hot plug (without reset) should done. I must admit, I did not care for this case.

pavelrevak, 2019-10-19:
Yes, Although in previous versions I don't have feature flashing under remapped SWD, the way was:
1. bring device into hard reset (also it is working with jumper or hardwired button)
2. sent command core_reset_halt
3. release hard reset
4. continue flashing normally

points 1 and 3 we can omit if we don't have remapped SWD signals.

I did not think about state when is CPU in deep sleep, but I think it will work with same way.

pavelrevak, 2019-10-19:
or do you think about cases when SW reset is not enough ?

Personally on many devices I don't connect for programmer HW reset signal (only SWDIO, SWCLK), because SW restart is in all my cases enough.