pavelrevak/pystlink #21: stm32l151zDT6 not supported. Check protection error

wolsty7, 2020-03-03:
running:
sudo python3 pystlink.py -d

returns:
Connected to ST-Link/  V2, serial 38ff6a064d43323542092243
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 28 80 83 04 48 37
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 01 00
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 97 06 00 00 00 08 00 00
  USB > f2 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 30 a3 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 77 14 a0 2b
DEVICE: ST-Link/V2 V2J34S0
SUPPLY: 2.91V
COREID: 2ba01477
Stm32.core_halt()
Halted after 0 transactions
  USB > f2 36 00 ed 00 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 30 c2 2f 41
CPUID:  412fc230
CORE:   CortexM3
  USB > f2 36 00 20 04 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 36 64 38 10
IDCODE: 10386436
  USB > f2 36 cc 00 f8 1f 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 00 00 00
*** Connected CPU with DEV_ID: 0x436 and FLASH size: 0KB is not supported. Check Protection ***
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
  USB > f2 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DONE in 0.14s

I get similar results with the stlink v3

wolsty7, 2020-03-03:
on the stm32L100RCT6 running sudo python3 pystlink.py -d
Connected to ST-Link/  V2, serial 38ff6a064d43323542092243
  USB > f1 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 28 80 83 04 48 37
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 00 01
  USB > f3 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB > f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 96 06 00 00 f5 07 00 00
  USB > f2 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 30 a3 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 80 00
  USB > f2 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 77 14 a0 2b
DEVICE: ST-Link/V2 V2J34S0
SUPPLY: 2.90V
COREID: 2ba01477
Stm32.core_halt()
Halted after 256 transactions
  USB > f2 36 00 ed 00 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 30 c2 2f 41
CPUID:  412fc230
CORE:   CortexM3
  USB > f2 36 00 20 04 e0 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 27 64 f8 10
IDCODE: 10f86427
  USB > f2 36 cc 00 f8 1f 00 00 00 00 00 00 00 00 00 00
  USB < 80 00 00 00 00 01 00 00
MCU:    STM32L100xC
FLASH:  256KB
SRAM:   16KB
EEPROM: 4KB
Stm32.core_nodebug()
  USB > f2 35 f0 ed 00 e0 00 00 5f a0 00 00 00 00 00 00
  USB < 80 00
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 02 00
  USB > f2 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB > f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  USB < 01 00
DONE in 0.35s

which look like it is working

wolsty7, 2020-03-03:
as a workaround I was able to get it to flash my device by setting the _flash_size to 384 if it was 0 after trying to get the size in find_mcus_by_flash_size()

so it seems it is having an issue detecting the flash size

pavelrevak, 2020-03-03:
Yes, this is bug, or nonstandard by ST, I see this, in [datasheet](https://www.st.com/content/ccc/resource/technical/document/reference_manual/cc/f9/93/b2/f0/82/42/57/CD00240193.pdf/files/CD00240193.pdf/jcr:content/translations/en.CD00240193.pdf) in section **31.1.1 Flash size register** is exception for devices with your dev_id `0x436`, where can be 0 for 384KB or 1 for 256KB of FLASH. other devices have on this address size of FLASH in KB

pavelrevak, 2020-03-03:
this need an workaround into for this dev_id

wolsty7, 2020-03-04:
Cheers mate, this project is awesome, I was seaching for a stlink programmer that would run on a 32bit raspberry pi for ages and nothing worked apart from this.